rust-embedded-community/embedded-sdmmc-rs #1: Make default busy-wait delay much shorter

birkenfeld, 2019-01-22:
Using a sdcard with a low-frequency STM32L0, the default 100000 spins was much too long, I reduced it to 100. (With that, everything works perfectly, thanks very much for the library!)

IMO if we're busy-waiting anyway, there shouldn't be a downside to polling more often also on faster CPUs?

thejpster, 2019-01-24:
> /// A terrible hack for busy-waiting the CPU while we wait for the card to
> /// sort itself out.
> ///
> /// @TODO replace this!

Yes, it's terrible! I'll try 100 on my board and if that works, push the change.

Glad to know it works on an STM32 - that's a great indication that Rust Embedded's HAL approach works! If you can, please do tweet @rust_embedded and let them know :)

thejpster, 2019-01-24:
PS: The delay was only there because I was worried that hammering the card as fast as we good made one of the cards I was testing with a bit upset.

birkenfeld, 2019-01-25:
That's perfect,thank you!