rust-embedded-community/embedded-sdmmc-rs #2: Consider using fat-rs

thejpster, 2019-02-19:
We should consider switching to FAT-rs (https://gitlab.com/susurrus/fat-rs/tree/master). No sense in duplicating work, if it:

* Works on `#[no_std]`
* Builds on stable
* Doesn't use alloc

thejpster, 2019-02-19:
(https://github.com/rafalh/rust-fatfs fails this test because it uses `core_io`, which needs nightly, and uses `alloc`)

twilco, 2019-02-20:
From `fat-rs`'s README:

> Eventually this will be migrated to use core_ios traits in order to be compatible with other libraries in the nascent no_std Rust ecosystem.

Looks like they might be moving towards `core_io`, too.  Also, perhaps this is an ignorant question, but is it odd that these crates are moving towards `core_io` which requires `alloc`?  Isn't the idea with embedded to avoid allocation whenever possible, meaning the utility of these crates becomes more limited by using implementations that require allocation?

Finally, I haven't been able to find a published crate on [crates.io](https://crates.io/search?q=fat) for `fat-rs` - perhaps I'm just missing it? 🙂

rafalh, 2019-06-12:
Please note https://github.com/rafalh/rust-fatfs requires `alloc` only if you need LFN support.

thejpster, 2019-10-31:
We have write support now. But we should keep an eye on what other crates are doing.

closed 2019-10-31 by thejpster