rust-embedded-community/embedded-sdmmc-rs #8: SdMmcSpi is not thread-safe

Lucus16, 2019-11-05:
As far as I can tell, it's possible that multiple immutable references are held to a single SdMmcSpi object and because the `read` and `write` methods accept `&self` rather than `&mut self`, they can be run concurrently using interrupt handlers or async/await. This will obviously not work when SPI bytes get mixed. I suggest making any method that requires SPI communication accept `&mut self` to ensure at compile time that access to it is limited to one at a time. I suggest doing the same for the methods of the `BlockDevice` trait as most implementations will require it.

Please let me know if I misunderstand something or if I should go ahead and create a PR for this.

thejpster, 2019-11-05:
Yeah, there's a reason it takes '&self'. I seem to recall it's a problem with the filesystem holding on to references to the disk controller, which means you can't have two filesystems on one disk.

thejpster, 2019-11-05:
A better workaround may be interior spin locks, or a mutex of some type `T` where `T: Mutex` and `Mutex` is the mutex API the embedded WG have been talking about.

Lucus16, 2019-11-17:
I think it would be better to pass a mutable reference to the disk controller explicitly to each function so the user of the library can choose whether they really need concurrency protected by a lock.

I should let you know that we've chosen not to use a filesystem for the job I'm working on, so I no longer have much of an interest in getting this fixed, but I still recommend either presenting a safe API or marking functions as explicitly unsafe.

thejpster, 2019-11-17:
So, because the type isn't entirely composed of types that are Sync (i.e. a reference to it being Send) then I don't think you can safely get a reference to it in two threads anyway. If there was a Mutex wrapped version, then that would be sync.

To me, it does seem that this library is sound as-is, but I'd be very interested in any examples demonstrating to the contrary. For now though, I am going to close this.

closed 2019-11-17 by thejpster