rust-embedded-community/embedded-sdmmc-rs #12: LFN support doesn't handle emoji

thejpster, 2019-11-08:
The LFN code assumes the code points are all valid UTF-16 and not surrogate pairs exist. This may not be a valid assumption. We should do a proper UTF-16 to UTF-8 conversion.

jonathanpallant, 2024-10-25:
See https://github.com/rust-embedded-community/embedded-sdmmc-rs/pull/157

thejpster, 2024-10-27:
I wanted to see how characters outside the BMP are encoded on Windows 11 on a FAT16 filesystem.

The file was called `Smiley 😀.txt`.

```text
[2024-10-27T11:47:34Z DEBUG embedded_sdmmc::fat::volume] LFN Contents true 1 90 [0053, 006d, 0069, 006c, 0065, 0079, 0020, d83d, de00, 002e, 0074, 0078, 0074]
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 't'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'x'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 't'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push '.'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push '?'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push '?'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push ' '
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'y'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'e'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'l'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'i'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'm'
[2024-10-27T11:47:34Z TRACE embedded_sdmmc::filesystem::filename] LFN push 'S'
```

We can see that `😀` (`\U{1F600}`) was encoded as `['\U{D83D}', '\U{DE00}']`, i.e. as a surrogate pair from the D800-DFFF range. Therefore to turn an LFN back into a String, we need to handle surrogate pairs.

thejpster, 2024-10-27:
Note that worst case, a file could consist of 255 code units, each between `\U{0800}` and `\U{D7FF}`. This would each be encoded as 3 bytes in UTF-8, meaning the worst-case size for an `LfnBuffer` is 765 bytes.