rust-embedded-community/embedded-sdmmc-rs #17: Remove dependencies on no_std

ceigel, 2020-03-03:
I discovered a strange behavior when I add embedded-sdmmc crate. I created a new embedded project (https://github.com/ceigel/sdmmc-test) with command cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart
I added the embedded-sdmmc crate and mfrc522 (this one just as control) as dependencies. The project compiles fine. When I add bindgen as a dev dependency, I get the following error:

`    Updating crates.io index
   Compiling byteorder v1.3.4
   Compiling log v0.4.8
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7m-none-eabi` target may not be installed

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: could not compile `log`.
warning: build failed, waiting for other jobs to finish...
error: build failed`

I wanted to use bindgen to generate bindings to a C++ library during compile time, but it seems that I need to do that part manually. 

To me this is very strange behavior and I wonder if it will break even after I remove my dev-dependency to bindgen.

thejpster, 2020-03-03:
Ah, welcome to the hell that is Cargo issue #5730. Features for dev dependencies get merged in with your build features. And bindgen includes some popular crates that have a 'std' flag.

The workaround is to `cargo install bindgen` and then call bindgen as a binary from your build.rs. Or use the very latest nightly as I think that might now have a fix.

thejpster, 2020-03-03:
Closing as it's not an embedded-sdmmc specific issue.

closed 2020-03-03 by thejpster

ceigel, 2020-03-03:
Thank you for the clarification.