rust-embedded-community/embedded-sdmmc-rs #18: Offer open-file type of function that receives a DirEntry

ceigel, 2020-04-11:
I am trying to iterate through a directory and open all files with a certain extension. There is no way to use the DirEntry received from iterate_dir as an input to open_file_in_dir. Ideally there would be a counterpart of open_file_in_dir which receives a DirEntry. One extra advantage of such a function is that it wouldn't be used with Mode::ReadWriteCreate, which means the volume can be a non-mutable reference.
Also the ShortFileName doesn't support retrieving the file extension.

thejpster, 2020-04-11:
Good spot! Very happy to take a PR on this, it sounds like a really good idea :)

ceigel, 2020-04-12:
Your wish is my command. Here is the PR: https://github.com/thejpster/embedded-sdmmc-rs/pull/19

ceigel, 2020-11-01:
Fixed

closed 2020-11-01 by ceigel