rust-embedded-community/embedded-sdmmc-rs #27: How do I keep a reference to the SPI peripheral?

Maldus512, 2020-11-12:
Hello,
I'm working on a custom embedded board with an SPI-interfaced microSD card. 
I've tried using this library to write logs on the card and it works using the `SdMmcSpi` interface. Unfortunately the SD card is not the only device with which I should communicate via SPI and the `SdMmcSpi` struct requires ownership of the `embedded_hal` peripheral object.

What this means for me is that I cannot have a `Controller` and communicate with other SPI devices in the same program. Is there a way to keep a reference to the SPI network after I'm done with the SD card?

I've noticed that there is a `spi` method for the `SdMmcSpi` struct that returns a mutable reference to the network, but it would be quite cumbersome (and counterintuitive) to keep the SPI reference inside the emmc `Controller` and pass it around to other devices.

Note that I'm far from an expert in Rust, so I might be missing something obvious here (I apologize if that's the case).

eldruin, 2020-11-13:
Hi @Maldus512, you can use [shared-bus](https://crates.io/crates/shared-bus) to share peripherals.

Maldus512, 2020-11-13:
Thank you, this is exactly what I meant by "missing something obvious".

closed 2020-11-13 by Maldus512