rust-embedded-community/embedded-sdmmc-rs #43: Suggestion: FAT filesystem impl as its own crate.

nihalpasham, 2021-12-29:
Firstly, thank you for this repo. It saved me a ton of time. 👍🏾 (I'm posting this as an issue as I couldn't find a discussions tab)

I'd like to make a suggestion - can we split the `FAT filesystem impl` into its own crate as the current implementation is pretty modular and it doesn't take a lot to turn it into a standalone crate. I wrote a [bare-metal EMMC driver](https://github.com/nihalpasham/rustBoot/blob/main/boards/test_impls/rpi4/src/bsp/drivers/sdhost.rs) for the `raspberry-pi 4` using `embedded-sdmmc-rs` (sans the SdMmcSpi controller). 
- **Reason:** rpi4 now has an `emmc2-controller` with dedicated pins connecting it to the on-board microSD slot and this slot isn't accessible via SPI. 

I think a standalone `no_std fat-fs` crate would be easier to work with when adding fat-support to block devices driven by other (non-spi) controllers.

MathiasKoch, 2022-03-10:
Just in case it was somehow missed, https://github.com/rafalh/rust-fatfs/ is no_std compatible and could perhaps be used eventually?

nihalpasham, 2022-03-19:
Good to know, thank you. This is what I was looking for.

closed 2025-07-20 by thejpster