rust-embedded-community/embedded-sdmmc-rs #44: log pulls in std: how do I disable the std feature on the log crate?

gauteh, 2022-01-05:
Optionally add `default-features = false` to log in you Cargo.toml.

Is this even possible?

gauteh, 2022-01-05:
Seems like something happened when going from log 0.3 to 0.4, it works with log 0.3.6.

eldruin, 2022-02-03:
@gauteh  Can this be solved by using `resolver = "2"` in the `Cargo.toml`?

gauteh, 2022-02-06:
Not sure, but with #45 this is no longer an issue for me.

Laura7089, 2022-03-07:
I'm having this issue as well.

> Related: https://github.com/rafalh/rust-fatfs/issues/39

Could it be that the `env_logger` dev dep is pulling in a `log` with `std` enabled?

Laura7089, 2022-03-07:
Might it be worth providing no-op implementation(s) of any `log` macros used for if the `log` and `defmt-log` features are both disabled?

gauteh, 2022-03-08:
Normally that is what `log` does if no logger is set up. It seems that `std` is pulled in because of dev/test dependencies (features). Might be solved by the suggestion of using `resolver = 2`. It is a bit more cumbersome to disable the statements completely.

I tried looking for a noop log crate, but this is already the standard log crate with no loggers.

PR #58 (2022-08-22): Fix env_logger dependency enabling std feature in log
One last fix before the 0.4.0 release

Fixes #44
closed 2022-08-22 by eldruin