rust-embedded-community/embedded-sdmmc-rs #47: confusion with directory documentation

x37v, 2022-02-01:
A couple notes/questions:

1. `open_root_dir` and `open_dir` both indicate: `Open a directory. You can then read the directory entries in a random order using get_directory_entry.` But `get_directory_entry` doesn't exist in the code AFAIK.  I see `find_directory_entry` and `iterate_dir` so I assume this must just be left over cruft?

2. `open_root_dir` and `open_dir` return a `Directory`, the docs for `close_dir` indicate:
`Close a directory. You cannot perform operations on an open directory and so must close it if you want to do something with it.` but `close_dir` consumes a directory so how do you use it after its closed? For instance, how would you supply a directory to `open_file_in_dir` ?

x37v, 2022-02-01:
looking at this example.. i'm guessing the `close_dir` documentation is wrong?

https://github.com/rust-embedded-community/embedded-sdmmc-rs/blob/ed1b7eaf5c559d7af3a3ea274049a677f18b1d8c/examples/test_mount.rs#L129-L144

x37v, 2022-02-21:
we resolved `1` with that PR but `2` .. the documentation about `close_dir` is still confusing

thejpster, 2022-07-06:
I think 2. means you can’t create a file in a directory when the directory is open? Because then you could get iterator invalidation if the entries moved around. But I agree the wording is awful.