rust-embedded-community/embedded-sdmmc-rs #52: Version release

ostenning, 2022-04-20:
@thejpster there hasn't been a release on crates.io for quite a long time. Some features on the HEAD, such as deletion of files is not accessible for those using the crate directly from the crates.io.

Is it possible to schedule a new release soon?

Kind regards

thejpster, 2022-04-26:
Good idea! Not sure when I’ll have time, but that’s why the crate lives here - so there’s others who can press the buttons as well.

ostenning, 2022-06-28:
> Good idea! Not sure when I’ll have time, but that’s why the crate lives here - so there’s others who can press the buttons as well.

I'm not entirely sure what the release etiquette surrounding project dependencies is. The issue I'm facing is that the [stm32h7xx_hal uses the 0.0.3 version from crates.io](https://github.com/stm32-rs/stm32h7xx-hal/blob/6698c1f423ef40d6eb8697b774ad923858d23e8a/Cargo.toml#L41) 

I can suggest to the stm32h7xx HAL maintainers to use this repository directly if that is what you recommend.

Thanks!

thejpster, 2022-07-06:
You can't publish to crates.io with a git dependency, so we should probably do a new release. Anyone in r-e-c should be able to do that - I have very limited time right now.

thejpster, 2022-07-06:
See #57

PR #57 (2022-07-10): Prepare 0.4.0 release
Summary:
- Bump version to 0.4.0
- Add custom error type for `File` operations currently with only one variant: `File::InvalidOffset`
- Split filesystem and fat module into submodules (no API changes)
- Extract `Controller` into submodule (no API changes)
- Small code improvements suggested by clippy
- Update env_logger dependency
- Update repository URL
- Remove Travis CI

Closes #52
closed 2022-07-10 by eldruin