rust-embedded-community/embedded-sdmmc-rs #55: Make `BlockDevice::read` and `BlockDevice::write` receive `&mut self`

datdenkikniet, 2022-06-11:
Hi,

Currently, `BlockDevice::read` and `BlockDevice::write` receive `&self`.

Why is this done? 

In my opinion, they should receive `&mut self`, so that responsibility of choosing what method of resource sharing a user wants is up to the user. Issues such as #37 could easily be avoided, and would also make more sense logically: `write`ing something is clearly modifying something, so why does it not need mutable access?

I'm willing to contribute a PR to make this change, if it is deemed that this makes sense.

datdenkikniet, 2022-06-13:
I also looked through some of the dependents on crates.io:

The only implementor of `BlockDevice` that I could find was in `stm32h7xx-hal` (the rest use it only in examples), and that uses the same `RefCell` hack to get around the non-mutability problem

datdenkikniet, 2022-06-18:
I'm working on an alternative implementation here:

https://github.com/datdenkikniet/embedded-sdmmc-rs/tree/no_more_immutable

eldruin, 2022-07-05:
Have you looked through the closed issues? There have been related discussions in the past.

eldruin, 2022-07-05:
Maybe @thejpster has some further reasoning for this interface choice?

datdenkikniet, 2022-07-05:
Yeah, that's what I was hoping for too :)

datdenkikniet, 2022-07-06:
I see, thanks for the info!

I'm cooking up a little something, and can ping again here once it's more or less done, and if it seems like the API can be on a similar level of user-friendliness.

So far I've finished implementing reading files and directories, but need to figure out how to do writing and reading at some point while staying ergonomic.

datdenkikniet, 2023-04-21:
I never did finish my work on this, and with #79 it may not make any sense at all :)

I'm closing this for now

closed 2023-04-21 by datdenkikniet