rust-embedded-community/embedded-sdmmc-rs #56: SdMmcSpi::acquire should consume self, not use &mut ref

ceigel, 2022-06-17:
Currently SdMmcSpi::acquire returns a BlockSpi object which contains a mutable reference to the SdMmcSpi. This is a construct that is very hard to work with. I'm reading parts of a file inside an interrupt, so I have to keep alive both a SdMmcSpi and a BlockSpi. Moreover the SdMmcSpi has to be placed in a static variable so that it is not moved once the BlockSpi is constructed. This is quite hard to use.
I believe it would be easier if the SdMmcSpi would be moved inside BlockSpi.

gauteh, 2022-10-01:
This would make it hard to re-acquire the BlockDeviceSPI in case of some error. I've wrapped the BlockDeviceSpi in a handle which makes it a bit easier to work with: https://github.com/gauteh/sfy/blob/main/sfy-buoy/src/storage/mod.rs#L204, depends on #63 .

thejpster, 2023-05-06:
Closed by #80

closed 2023-05-20 by thejpster