rust-embedded-community/embedded-sdmmc-rs #66: File `starting_cluster` has a value of zero within the `open_files`

ostenning, 2022-09-30:
I've found an issue when trying to close files and open them again. Basically what ends up happening is that upon opening the file, the [`starting_cluster` of the File initialized to zero](https://github.com/rust-embedded-community/embedded-sdmmc-rs/blob/develop/src/controller.rs#L365), and then this zero value [is subsequently set on the `open_files`](https://github.com/rust-embedded-community/embedded-sdmmc-rs/blob/develop/src/controller.rs#L373).

At some point the `starting_cluster` value of the File is adjusted, but the value stored within `open_files` is not. 

Which in turn causes the [`close_file` function within the controller](https://github.com/rust-embedded-community/embedded-sdmmc-rs/blob/develop/src/controller.rs#L576) to not close the file.

If you attempt to open/close files multiple times (beyond the specified `MAX_FILES`) value, the library returns an Error.

I'm trying to figure out a solution to this at the moment. Any help is appreciated. Thanks

jbeaurivage, 2022-10-17:
I can second that this is happening to me too. It also seems the same thing happens when attempting to close directories.

thejpster, 2022-10-18:
Oh no. Does the volume this happens on pass a CHKDSK? I didn't think you could have a directory entry pointing at a cluster ID of zero. But I guess anything that /can/ happen on-disk /could/ happen so we should be robust to it.

PR #67 (2023-06-11): Fix closing `File`s and `Directories`
This PR attempts to fix closing `Files` and `Directories`.

The working principle is that each new `File` and `Directory` is assigned a unique `SearchId`. To close/delete them, instead of looking for magic `Cluster` numbers, we simply need to lookup the ID in the list of open files or directories. This mitigates the issues where we might forget to update the cluster number, or where there might be multiple cluster numbers with the same value (most notably, 0).

Also replace raw arrays with `heapless::Vec`s for a cleaner implementation.

Optionally, this co [... cut]
closed 2023-06-11 by thejpster