rust-embedded-community/embedded-sdmmc-rs #68: interoperability with esp_rs

DS3a, 2023-01-09:
I could not find any information about the microcontrollers I can use this crate with. The other repositories in this organization seem to be aimed at cortex-m series microcontrollers. Can I use this with esp32 using the `esp_rs` crate?

elpiel, 2023-02-25:
Yes, you can. the esp32 hal crates all implement `embedded_hal` traits which are compatible with `embedded_sdmmc`:
https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/spi.rs

DS3a, 2023-02-26:
Alright, thanks for confirming

closed 2023-02-26 by DS3a

niuhuan, 2023-05-28:
Has any code snippets for me to learn? 

I used embedded_svc,  embedded_hal, esp_idf_svc, esp_idf_hal, but I don't know how to get the parameters sdmmc_spi, sdmmc_cs, delay.

I hope to provide me with a link to take a look at the ESP code that uses this crate. Thank you for your help

```rust
    let spi = esp_idf_hal::spi::Spi::device();
    let sc = esp_idf_hal::spi::SpiConfig::default();
    let sdcard = embedded_sdmmc::SdCard::new(sdmmc_spi, sdmmc_cs, delay);
```

[image]

niuhuan, 2023-05-30:
```rust
    let mut spi_d_config = esp_idf_hal::spi::config::DriverConfig::default();
    let spi_d = esp_idf_hal::spi::SpiDriver::new(
        peripherals.spi3,
        peripherals.pins.gpio18, // sclk
        peripherals.pins.gpio23, // sdo
        None::<Gpio0>,           // sdi
        &spi_d_config,
    )
    .expect("panic spi_d");
    let mut spi_dd_config = esp_idf_hal::spi::SpiConfig {
        baudrate: esp_idf_hal::units::Hertz(1000 * 1000),
        ..Default::default()
    };
    let sd_spi_dd = esp_idf_hal::spi::SpiDeviceDriver::new(
        &spi_d,
        Some(peripherals.pins.gpio4),
        &spi_dd_config,
    )?;
    let sdcard = SdCard::new(
        sd_spi_dd,
        PinDriver::output(peripherals.pins.gpio19)?,
        SdDelayer,
    );
    info!("sdcard init : {:?}", sdcard.get_card_type());
    info!("sdcard init : {:?}", sdcard.num_bytes());
    let mut volume_mgr = embedded_sdmmc::VolumeManager::new(sdcard, FakeTimesource {});
```

but ....

```
W (4729) embedded_sdmmc::sdcard: Got response: 0, trying again..
W (4739) embedded_sdmmc::sdcard: Got response: 0, trying again..
W (4739) embedded_sdmmc::sdcard: Got response: 0, trying again..
W (4749) embedded_sdmmc::sdcard: Got response: 0, trying again..
W (4759) embedded_sdmmc::sdcard: Got response: 0, trying again..
W
```

jonathanpallant, 2023-05-30:
I don't know how ESP-IDF works, but the pin you give to the `SdCard` constructor should be DAT3/CD - so GPIO4 according to your picture. Not giving a pin for SDI in the SPI constructor also seems odd.

In any case, we can't give MCU specific instructions here otherwise we'd be spending forever looking at different MCUs. Maybe the ESP-RS team and put together an example, like the RP-RS team did for the Raspberry Pi Pico.

niuhuan, 2023-06-01:
I have any micro python is run ok.

```python
class SDCard:

    def __init__(self, spi=SoftSPI(sck=Pin(18), mosi=Pin(23), miso=Pin(19)), cs=Pin(4), debug=False):
        """
        get ready to read/write, the default Pins are set up for M5Stack Core2 hardware
        """
        self.spi = spi
        self.cs = cs
        self.cs.init(self.cs.OUT, value=0)
        self.debug = debug
        if self.debug:
            print("{}{} {}, cs={} state={}".format(SDCard, SDCard.__init__, self.spi, self.cs, self.cs.value()))
        self.sectors = None
        self.cdv = None
        self.init_card()

class SDCardTest:
    """ run large and small files read/write and delete tests """

    def __init__(self, mdir='/sd', debug=False):
        self.mdir = mdir
        self.debug = debug
        self.run()

    def run(self):
        sdc = SDCard(debug=self.debug)
        vfs = os.VfsFat(sdc)
        os.mount(vfs, self.mdir)

```

niuhuan, 2023-06-01:
i am tried this, but not run ok

```rust
fn main(){
    let mut spi_d_config = esp_idf_hal::spi::config::DriverConfig::default();
    let spi_d = esp_idf_hal::spi::SpiDriver::new(
        peripherals.spi2,
        peripherals.pins.gpio18,       // sclk
        peripherals.pins.gpio23,       // sdo
        Some(peripherals.pins.gpio19), // sdi
        &spi_d_config,
    )
    .expect("panic spi_d");

    let mut spi_dd_config = esp_idf_hal::spi::SpiConfig::default();
    spi_dd_config.baudrate = Hertz(24_000_000);
    spi_dd_config.duplex = Duplex::Full;

    let sd_spi_dd = esp_idf_hal::spi::SpiDeviceDriver::new(&spi_d, None::<Gpio0>, &spi_dd_config)?;
    let sdcard = SdCard::new(
        sd_spi_dd,
        PinDriver::output(peripherals.pins.gpio4)?,
        SdDelayer,
    );
    info!("sdcard init : {:?}", sdcard.get_card_type());
    let mut volume_mgr = embedded_sdmmc::VolumeManager::new(sdcard, FakeTimesource {});
    let mut volume0 = volume_mgr.get_volume(embedded_sdmmc::VolumeIdx(0)).unwrap();

}
```

jonathanpallant, 2023-06-02:
You usually can't do card init at full speed - you have to drop to 400 kHz to do the init, and once the card has responded, then you can go back to full speed.

niuhuan, 2023-06-02:
Good job , thank you. scard is work.