rust-embedded-community/embedded-sdmmc-rs #79: Explore interior mutability

jonathanpallant, 2023-04-21:
Now that `critical-section` is a thing, maybe we can consider interior mutability:

* `Volume` can hold an `&VolumeManager`
* `Directory` and `File` can hold an `&Volume`

```rust
let volume = volume_mgr.get_volume(VolumeIdx(0))?;
let root_dir = volume.open_root_dir()?;
let file = root_dir.open("HELLO.TXT")?;
let mut buffer = [0u8; 64];
let contents = file.read_to_buffer(&mut buffer)?;
```

thejpster, 2023-05-20:
So one issue here is stopping people opening the same volume twice, but allowing all the objects to be stashed in a struct for later use (so they can't refer to each other).

Perhaps the VolumeManager can hold all the volumes internally and lend out &mut Volume references cheaply, but only for one Volume at a time. You could even const generic the number of Volumes a VolumeManager can hold (one is usually fine, but sometimes there's an EFI partition first or something). Then I think for opening files you can either get a File (which holds a &Volume) or a RawFile (which does not, but does hold some unique ID for the Volume so it'll panic if you use it with the wrong Volume in future).

thejpster, 2023-10-07:
Looked at this in https://github.com/rust-embedded-community/embedded-sdmmc-rs/pull/103 is a start on this, but using `&mut` to start.

Also there's an old branch: https://github.com/rust-embedded-community/embedded-sdmmc-rs/tree/make_api_immutable

thejpster, 2024-10-12:
Closed by #143

PR #143 (2024-10-12): Use a RefCell to provide interior mutability for VolumeManager.
VolumeManager now contains a RefCell, so all the methods can be `&self`. This also means you can have multiple smart handles usable at the same time.

Note that VolumeManager is !Sync, so this is just about having multiple `File` objects active at one time.

Closes #79 (finally!)
closed 2024-10-12 by thejpster