rust-embedded-community/embedded-sdmmc-rs #82: State::Idle should be State::Initialized

elpiel, 2023-05-07:
### Idle vs Initialized state

Accordingly to this answer on StackExchange, the current `State::Idle` that we set after the initialisation process is actually an Initialised state. 

https://electronics.stackexchange.com/a/602106

### Clock initialise speed

It also seems that this initialisation process should be made in 100 - 400 kHz SPI clock speed (`acquire`) and should later be increased to the running clock speed.

I suggest that we at least include this in the documentation and later improve (if possible) the crate to handle the clock speed.
Since there isn't any `Config` for the spi interface in `embedded-hal` (0.2) I see documentation as the only option for now.

jonathanpallant, 2023-05-08:
Yes we should document the clock speed requirements if it's not currently clear. Happy to take a PR which changes #80.

thejpster, 2023-05-20:
Replaced with an Option<CardType> as it's either initialised or it isn't.

closed 2023-05-20 by thejpster