rust-embedded-community/embedded-sdmmc-rs #87: Acquire SD card

elpiel, 2023-05-24:
I've come to realise that in version `0.5` in order to acquire the SD card you need to call `check_init` or `acquire` but it's not exposed from `SdCardInner`.
Instead you need to call either `num_bytes` or directly try to read/write to the SD card in order to initialize it.
It would be nice to be able to expose `acquire` or better the `check_init` function in `SdCard` in order to be able to initialize it upfront before the first read/write takes place.

jonathanpallant, 2023-05-24:
In theory *any* API call will trigger card acquisition. The demos just use `num_bytes()` because it's a nice one to print out.

elpiel, 2023-05-24:
Yes, it would be nice to have it also in the card type method.
My only shorts the current acquiring method is that it is implicit.
This needs to be well-documented or have a separate method to initialize the card imo.

I've tested today, I can say how much improvement there is, however, new version is working 🙌

jonathanpallant, 2023-05-25:
It's a replacement for the Arduino SDFat library, so keeping it simple is good.

elpiel, 2023-06-21:
@jonathanpallant should I close this issue or should I leave it because of the `get_card_type()`?

jonathanpallant, 2023-06-21:
I opened https://github.com/rust-embedded-community/embedded-sdmmc-rs/issues/90

elpiel, 2023-06-21:
Ok, great!
Then we can either close this issue or set it to "documentation needed". I still believe it should be mentioned that it acquires the sd card on first call in the docs.

commit 3aa510f7 (2023-09-15): Clarify what inits that card and what doesn't.

Closes #87
Closes #90
PR #95 (2023-09-23): Cleaning up some comments, and fixing the close file bug.
Whilst I was tidying up, I found #94. Let's fix it!

Also fixes #87 and #90
closed 2023-09-23 by thejpster