rust-embedded-community/embedded-sdmmc-rs #96: Do a release

jonathanpallant, 2023-09-15:
Once #95 is in, we should do a breaking change release.

thejpster, 2023-09-16:
`cargo semver-checks` says:

*  field AcquireOpts.acquire_retries in C:\Users\msn\Documents\GitHub\embedded-sdmmc-rs\src\sdcard\mod.rs:633
   * Adding a pub field to a literal constructable struct is a breaking change
* struct embedded_sdmmc::Controller, previously in file C:\Users\msn\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embedded-sdmmc-0.5.0\src\volume_mgr.rs:16
   * I removed the Controller alias

So it will be 0.6.0

thejpster, 2023-11-03:
Now we need to do 0.7 with the smart handles.

thejpster, 2023-12-17:
and the mkdir support.

bsodmike, 2024-02-25:
I noticed these have been merged, may be close this issue? @thejpster 😅

closed 2024-02-25 by thejpster