rust-embedded-community/embedded-sdmmc-rs #98: Add a flush API

thejpster, 2023-09-22:
An open file is risky because the data is still in RAM. Add a flush API which is like a close and a reopen.

peterkrull, 2024-03-24:
I just started using this crate. I intend to use it for high-rate logging on an aerial drone, where the probability of a crash and the log files being needed are highly correlated. So ensuring data is saved, even in the event the SD card is flung loose or instant loss of power, would be nice. I presume this flushing API would be a way to accomplish that?

thejpster, 2024-03-24:
Yes. It would be like closing the file and re-opening it. Basically you just need to update the directory entry to have the correct file size.

But also that sounds like a use-case where your data is important to you, and I only really wrote this crate to suit my own whimsical needs - so please do your own testing and verification before trusting it with anything.

peterkrull, 2024-03-24:
Thanks for the quick response. I briefly tried experimenting with dropping the `File` handle and acquiring a new one with `Directory::open_file_in_dir(..)` to force it to update the directory entry. This caused a significant slowdown in throughput if it was done too often, but it did like I was more likely to have a file if I yanked the SD card while my program was running. I will keeping testing and looking around to figure out what I need, but the ability to flush would definitely be a nice feature!

closed 2024-03-25 by thejpster