rust-embedded-community/embedded-sdmmc-rs #101: File.current_offset is private to the crate and there is no tell()

Jajcus, 2023-09-28:
The File object provides handy ``seek_*()`` methods to move to a specific position, but there is no ``tell()`` method to get the current position. This forces client applications to track file offsets themselves or use ``f.size() - f.left()``.

jonathanpallant, 2023-09-28:
Oops. Feel like sending a PR?

Jajcus, 2023-09-29:
Sure! I guess even Rust newbie like me can handle this one. :-)

Jajcus, 2023-09-29:
Oh… current git version is different than the released on – there are no File methods at all.

I guess this has to wait until  #99 is resolved.

Jajcus, 2023-09-29:
Oh… there already is VolumeManager.file_offset() method. So I guess that has been already solved.

thejpster, 2023-09-30:
Oh, nice. I'd forgotten about that.

closed 2023-09-30 by Jajcus