rust-embedded-community/embedded-sdmmc-rs #117: Reason for MS-DOS 8.3 filename compability

Murmele, 2024-02-08:
I tried to create a file with a longer filename and got the result of a too long filename. I have seen in the code, that the filename is limited to 8 characters + 3 extension. What is the reason for?

jonathanpallant, 2024-02-08:
That's how MS-DOS works, what FAT natively supports, and it's all I need.

You could add support for LFNs, but you'll have to work out how to store the string because it could be up to 255 UCS-2 characters long (or up to 1020 bytes as UTF-8), and that's a bit big to be passing around on the stack.

jonathanpallant, 2024-02-08:
(LFNs are stored as a series of directory entries, where the 16-bit UCS-2 characters are stored all across the 32-byte directory entry, trashing the timestamp and other fields. They are 'hidden' from normal accesses by virtue of being marked as READ ONLY, HIDDEN, SYSTEM, and VOLUME LABEL).

Murmele, 2024-02-10:
Thanks for the feedback. I will check out the options and will come back if I found a solution. Thanks for the information about the system behind, so I know a little bit more where to search for.

closed 2024-02-10 by Murmele