rust-embedded-community/embedded-sdmmc-rs #122: In case of early ejection of SD card, allow "closing" `RawFile` handle anyway

peterkrull, 2024-03-25:
It seems that there is no nice way to get rid of a RawFile handle if for example the SD card is ejected, since calls to `crate::VolumeManager::close_file` will result in a returned error before the file id is removed from the `open_files` list.

In my testing, I have implemented the following:

```rust
if volume_mgr.write(file, data.as_slice()).is_err() {
    volume_mgr.destroy_file(file).unwrap();
    volume_mgr.close_dir(dir).unwrap();
    volume_mgr.close_volume(volume).unwrap();

    continue 'setup; // Loop back and retry connecting to card
}
```

where 
```rust
/// Destroy a file for the given RawFile, without updating its entry.
/// This can be useful if the SD card is suddenly ejected, and one must
/// notify the VolumeManager that we are dropping the handle.
pub fn destroy_file(&mut self, file: RawFile) -> Result<(), Error<D::Error>> {
    let file_idx = self.get_index(file)?;
    self.open_files.swap_remove(file_idx);
    Ok(())
}

/// Retrieve the index for the given RawFile
pub fn get_index(&mut self, file: RawFile) -> Result<usize, Error<D::Error>> {
    let mut found_idx = None;
    for (idx, info) in self.open_files.iter().enumerate() {
        if file == info.file_id {
            found_idx = Some(idx);
            break;
        }
    }
    found_idx.ok_or_else(|| Error::BadHandle)
}
```
Which works nicely. Another option would be to modify the `close_file` method to first get the index, remove from the list, and then try to update the entry by flusing:

```rust
/// Close a file with the given full path.
pub fn close_file(&mut self, file: RawFile) -> Result<(), Error<D::Error>> {
    let file_idx = self.get_index(file)?;
    self.open_files.swap_remove(file_idx);
    self.flush(file)?; // or flush
    Ok(())
}
```

thejpster, 2024-03-25:
I think `close` should always close a file, regardless of whether it was able to flush it.

peterkrull, 2024-03-25:
I think the simplest solution is to simply not short-circuit the file-flush, then remove the file index, then return the result of the file-flush

peterkrull, 2024-03-27:
Closing in favor of #123

closed 2024-03-27 by peterkrull