rust-embedded-community/embedded-sdmmc-rs #129: warning: field `0` is never read

orsinium, 2024-04-16:
There are several warnings from clippy when running `cargo test` for the project:

```
warning: field `0` is never read
  --> examples/readme_test.rs:80:16
   |
80 |     Filesystem(embedded_sdmmc::Error<embedded_sdmmc::SdCardError>),
   |     ---------- ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |     |
   |     field in this variant
   |
   = note: `#[warn(dead_code)]` on by default
help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
   |
80 |     Filesystem(()),
   |                ~~

warning: field `0` is never read
  --> examples/readme_test.rs:81:10
   |
81 |     Disk(embedded_sdmmc::SdCardError),
   |     ---- ^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |     |
   |     field in this variant
   |
help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
   |
81 |     Disk(()),
   |          ~~

warning: field `0` is never read
  --> tests/utils/mod.rs:46:8
   |
46 |     Io(std::io::Error),
   |     -- ^^^^^^^^^^^^^^
   |     |
   |     field in this variant
   |
   = note: `#[warn(dead_code)]` on by default
help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
   |
46 |     Io(()),
   |        ~~

warning: field `0` is never read
  --> tests/utils/mod.rs:48:12
   |
48 |     Decode(flate2::DecompressError),
   |     ------ ^^^^^^^^^^^^^^^^^^^^^^^
   |     |
   |     field in this variant
   |
help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
   |
48 |     Decode(()),
   |            ~~

warning: field `0` is never read
  --> tests/utils/mod.rs:50:17
   |
50 |     OutOfBounds(BlockIdx),
   |     ----------- ^^^^^^^^
   |     |
   |     field in this variant
   |
help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
   |
50 |     OutOfBounds(()),
   |                 ~~

warning: `embedded-sdmmc` (example "readme_test") generated 2 warnings
warning: `embedded-sdmmc` (test "open_files") generated 3 warnings (3 duplicates)
warning: `embedded-sdmmc` (test "write_file") generated 3 warnings
warning: `embedded-sdmmc` (test "volume") generated 3 warnings (3 duplicates)
warning: `embedded-sdmmc` (test "directories") generated 3 warnings (3 duplicates)
warning: `embedded-sdmmc` (test "read_file") generated 3 warnings (3 duplicates)

```

thejpster, 2024-06-01:
Do you have special clippy rules or perhaps are you using nightly? I can't work out how to get some of those errors.

orsinium, 2024-06-02:
> or perhaps are you using nightly

Turns out I do!

```
$ cargo --version
cargo 1.79.0-nightly (74fd5bc73 2024-04-10)

$ rustc --version
rustc 1.79.0-nightly (a07f3eb43 2024-04-11)
```

thejpster, 2024-06-02:
Ok so for the test code I think I will just allow unused values, because they might be useful debug info if a test fails and I'm not sure the unused analysis pays attention to debug impls.

PR #134 (2024-06-06): Fix warnings
Cleans up some clippy warnings

Closes #129
closed 2024-06-06 by eldruin