rust-embedded-community/embedded-sdmmc-rs #131: `find_data_on_disk()` panics due to attempt to subtract with overflow

vinsynth, 2024-05-24:
Here's the relevant code (slight modification of `examples/read_file.rs`):

```
const FILE_TO_READ: &str = "64MB.DAT";

fn main() -> Result<(), embedded_sdmmc::Error<std::io::Error>> {
    env_logger::init();
    let mut args = std::env::args().skip(1);
    let filename = args.next().unwrap_or_else(|| "./disk.img".into());
    let print_blocks = args.find(|x| x == "-v").map(|_| true).unwrap_or(false);
    let lbd = LinuxBlockDevice::new(filename, print_blocks).map_err(Error::DeviceError)?;
    let mut volume_mgr: VolumeManager<LinuxBlockDevice, Clock, 8, 8, 4> =
        VolumeManager::new_with_limits(lbd, Clock, 0xAA00_0000);
    let mut volume = volume_mgr.open_volume(VolumeIdx(0))?;
    let mut root_dir = volume.open_root_dir()?;
    println!("\nReading file {}...", FILE_TO_READ);
    let mut f = root_dir.open_file_in_dir(FILE_TO_READ, Mode::ReadOnly)?;
    f.seek_from_start(0x2c)?;
    while f.offset() < 1000000 {
        let mut buffer = [0u8; 2048];
        println!("\npre-read offset: {}", f.offset());
        f.read(&mut buffer)?;
        println!("post-read offset: {}", f.offset());
        f.seek_from_current(-1024)?;
    }
    println!("Done!");
    Ok(())
}
```

...and the output (with backtrace):

```
Reading file 64MB.DAT...

pre-read offset: 44
post-read offset: 2092

pre-read offset: 1068
thread 'main' panicked at /home/vincent/Dev/linux/embedded-sdmmc-rs/src/volume_mgr.rs:1057:35:
attempt to subtract with overflow
stack backtrace:
   0: rust_begin_unwind
             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/panicking.rs:647:5
   1: core::panicking::panic_fmt
             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/panicking.rs:72:14
   2: core::panicking::panic
             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/panicking.rs:144:5
   3: embedded_sdmmc::volume_mgr::VolumeManager<D,T,_,_,_>::find_data_on_disk
             at ./src/volume_mgr.rs:1057:35
   4: embedded_sdmmc::volume_mgr::VolumeManager<D,T,_,_,_>::read
             at ./src/volume_mgr.rs:624:58
   5: embedded_sdmmc::filesystem::files::File<D,T,_,_,_>::read
             at ./src/filesystem/files.rs:78:9
   6: read_file::main
             at ./examples/read_file.rs:60:9
   7: core::ops::function::FnOnce::call_once
             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/ops/function.rs:250:5
```

The final offset reached before panicking varies depending on the negative seek amount for `f.seek_from_current()` and the buffer size; however, panicking always eventually occurs when the while loop is preceded by `f.seek_from_start()` with a seek amount that doesn't align with the buffer size.

thejpster, 2024-05-25:
Fascinating! Thank you for reporting this, and with a Linux based reproducer. Do you have any idea where it's going wrong? If not, I'll try and look at it late next week.

I guess I should have a test case that really exercises the seek logic.

thejpster, 2024-06-02:
I think I fixed this. It was definitely a bug on my part!

PR #133 (2024-06-10): Fix seeking
Hopefully closes #131. Now we start from the beginning of a file when the current offset is ahread of the desired offset.
closed 2024-06-10 by eldruin