rust-embedded-community/embedded-sdmmc-rs #135: Accepting `Read + Write + Seek`

felipebalbi, 2024-06-16:
Hi,

I understand that `Read`, `Write`, and `Seek` are part of `std::io`, but is there anything preventing those traits from moving into `core`? That way, `embedded-sdmmc` could implement those three traits for `File` and we can easily build abstractions on top that just take `Read + Write + Seek`, either from `std` (useful for testing, for example) or from `embedded-sdmmc`.

Sure, the change to `std::io` would be somewhat extensive:

1. Move traits to `core`
2. Produce a new `core::result::Result`
3. Map `core::result::Result` to `std::io::Result`
4. Update `File` implementation
5. New release...

But I think the benefits are actually tangible. In my example, I'm wrapping my `File` access in a crate of its own such that I could use it on my development machine or on embedded. When using with embedded, I wanted to use `embedded-sdmmc`'s `File`, otherwise `std::io`'s `File`.

thejpster, 2024-06-16:
That's a question for the Rust Project, but I think the issue is they rely on std::io::Error, which ... uses boxed dyn trait objects ...? Or it uses something that stops you putting it in core. I think they are working on it though.

thejpster, 2024-06-16:
I'm closing this as there's nothing I can do here until stuff happens upstream in libcore. But if such traits become available I'm very happy to look at adding them!

closed 2024-06-16 by thejpster