rust-embedded-community/embedded-sdmmc-rs #141: Support for formating SD card?

rowanfr, 2024-08-13:
I want to know if I've overlooked any functionality for formatting SD cards or if it's planned in the future. I can only seem to find information on how to print out information about them already preconfigured in FAT16 or FAT32 and the first line on the README being "This crate is intended to allow you to read/write files on a FAT formatted SD card on your Rust Embedded device" indicates that this crate is only for reading and writing not formatting.

If formatting is outside the scope of this crate then fair enough, but I want to double check.

thejpster, 2024-08-14:
Actually formatting and partitioning would be useful and I’d happily merge support for that.

rowanfr, 2024-08-14:
I'll look into implementing it then (though I wouldn't expect this soon as I'm not familiar with how SD cards work at a low level so it will require some research).

thejpster, 2024-08-14:
The Wikipedia article on the FAT format isn’t a bad place to start.

https://en.m.wikipedia.org/wiki/Design_of_the_FAT_file_system