rust-embedded-community/embedded-sdmmc-rs #156: Micro SD write speed on RP2040 is too slow

axoulc, 2024-10-25:
Hello,

Here is my configuration: 
- MCU : RP2040
- SPI speed: 16MHz
- Periodic write size: ~80 bytes

I'm using the `embedded-sdmmc-rs` library (version 0.5.0 because I can't get `SpiDevice` to work on `thumbv6m-none-eabi` architecture, although I've tried with `embedded-hal-bus` without success).
My program writes one line of a CSV file as quickly as possible and starts again. Each time I open the same file, prepare my bytes, write to the micro SD and close the file. I measured a time of around 54ms per write, which is far too long for my application (I'd like at least 10x faster).

Do you have any advice for my problem or even solutions?

Thanks in advance

michaelkamprath, 2024-10-25:
If you call the `VolumeManager::write` method for each individual byte, then that is a very slow process. Even if you write a small buffer, the overhead of everything that needs to happen just to lay down the bytes is high. I've written a buffer manager that wraps the `VolumeManager` for a specific file, and gives me a `write` like interface that instead of writing the byes to the SD card right away, they get written to a buffer that in turn writes to the SD card when sufficient bytes have been accumulated to make the overhead of the `write` worth it. Also added a `flush` method just to force the write if warranted.

BTW, I do have the latest version of `SpiDevice` working on the RP2040 just fine with the lates `embedded-hal`. You do need to implement the `transaction` method of the  `SpiDevice` trait. Don't forget that the chip select pin for a SPI device is active low ... that issue tripped me up for a while.

axoulc, 2024-10-25:
For the `SpiDevice`, I was just too lazy to develop this trait.
I'll study your answer. If you can share a snippet of your code for the `VolumeManager`, I'd love to.
Thanks in advance.
PS: and yes, Chip Select is written as /CS because it's active when low ;)

michaelkamprath, 2024-10-25:
Here is some sample code that I have working on a rp2040: https://github.com/michaelkamprath/rp2040-rust-robot/blob/main/src/robot/file_storage/sd_card_spi_device.rs