rust-embedded-community/embedded-sdmmc-rs #172: How to Create, then continuously Append

zpg6, 2025-01-17:
`Mode` enum is not very clear for how to operate when you plan to do several operations on a file. 

In my use case, I want to create a file and then continue to append data to it. When I use `ReadWriteCreateOrAppend`, the create works and it does not fail but the appending operations don't result in the file being modified.

Am I missing something?

thejpster, 2025-01-17:
I don't think I've ever tested that. Could you write a test case for me (that fails) in a PR so I can take a look?

zpg6, 2025-01-18:
UPDATE: solution was to make sure you `file.flush()`. For a device like mine where power could turn off at any time, I just flush after each write and its fine.

closed 2025-01-18 by zpg6