rust-embedded-community/embedded-sdmmc-rs #181: Needing to remove/insert sdcard before I can open it

nebogeo, 2025-02-27:
I am not sure if this is an issue with this library (I'm using the 0.8.1 version of the embedded-sdmmc crate) but I'm having an issue where everything works fine - but only if I remove and insert an sdcard after booting (I'm using a Raspberry Pi Pico with PiCowbell Adalogger).

Where should I be looking to solve this issue? I've tried messing around with mark_card_uninit() but no joy.

thejpster, 2025-02-27:
I would start with putting a logic analyser on the bus and capturing all the traffic, ideally along with an analog capture of the SD card power rail.

A defmt trace from the library would also help.

nebogeo, 2025-02-28:
Good advice, thanks - I only have a 2 channel handheld oscilloscope, but I can at least record what's happening.

I'll also try the Adafruit micropython version and do this same (I should have tried that before posting really).

I'll compare and report back.

nebogeo, 2025-06-02:
Apologies for the late response - this turned out to be a conflict with another device sharing the same SPI, so nothing to do with this driver. Initialising, opening and writing to a file before initialising the other device fixed this issue for me.

closed 2025-06-02 by nebogeo