rust-embedded-community/embedded-sdmmc-rs #188: Bug? Can't Seek to offset 0 from start and write - file gets corrupted

yanshay, 2025-05-30:
I'm writing to a file in chunks of 128 bytes. Every write is open file, write, flush and close of the file. Each such write I'll refer to as a transaction.

After writing three such transactions, the file size is 384 bytes.

Then
- If I seek to position 128 and write 128 bytes - write is ok, and those 128 bytes are modified.
- If I seek to position 256 and write 128 bytes - write is ok, and those 128 bytes are modified.
- If I seek to position 0 and write 128 bytes - write succeeds, but file is corrupted. First 128 bytes are as expected, the following 256 are zeros.
- If I don't seek it writes at the end of the file, so currently I'm unable to write to the beginning of the file.

I'm using the async (using bisync api), but I don't think it has anything to do with that.

Is it a bug or maybe something I could be doing wrong?

Some technical details in case it's relevent.
I create the file using `ReadWriteCreateOrAppend` and then open using `ReadWriteAppend`.
I use `seek_from_start`, `write`, `flush`, `close` - everything pretty standard.

thejpster, 2025-05-31:
Away at the moment, but sounds plausible. I'd love to see a PR with a test case that trips the bug (and hence fails) followed by a commit that demonstrably fixes it. The PR should be on the main branch - we can rebase the async stuff after.

thejpster, 2025-06-07:
I commented in the PR, but for anyone who finds this issue, my proposed fix is:

```rust
let block = if (block_offset == 0) && (to_copy == Block::LEN) {
    // we're replacing the whole Block, so the previous contents
    // are irrelevant
    data.block_cache.blank_mut(block_idx)
} else {
    debug!("Reading for partial block write");
    data.block_cache
        .read_mut(block_idx)
        .map_err(Error::DeviceError)?
};
```

i.e. always read the current contents of a block unless we know we are replacing the whole block anyway.

thejpster, 2025-06-07:
Closed by #189