rust-embedded-community/embedded-sdmmc-rs #206: How to read from and write to files under the same directory?

yanshay, 2025-09-27:
I want to read content from a file in the root folder and write content from there to files in the same volume at different levels of the folder hierartchy (so under the same root directory).

For that if I get it right, I need to open the root folder, open the file, then change directory and open file for write and write.

However, once the file is open (and I don't close it since I'm processing it), I can't change_dir (can't borrow the dir as mutable because it's borrowed for the open file).

So how can this be done?

Should I open a second instance of the root folder and change from there? 
What if I close the root folder of the second instance? will it close also the first? Or are these just two instances of the same folder that work separately? But then what happens if in one I add files? Will the other reflect that?

thejpster, 2025-09-28:
I would do what the [`hexdump()`](https://github.com/rust-embedded-community/embedded-sdmmc-rs/blob/c3aef4ad3c9ce91a9ab1cc8216fc8421e68df446/examples/shell.rs#L359) function does in the examples. It takes a path, opens the folders one at a time to walk the path, and returns the file handle. You can then do what you want with the file, and close it again.

thejpster, 2025-09-28:
Previous versions had File and RawFile, and File would mutably borrow the Volume Manager so you could only open one File at once (but you could have many RawFiles open, they just needed to be passed back to methods on the Volume Manager to use them).